pellcorp/grumpyscreen #25: Display chamber temp

Torstein-11825, 2025-02-18:
Would be nice to have chamber temp showing in print status window. This is for me relevant info during printing. 
There is also room for one more box below - the fan info box, where this info could be placed.
 [image]

Torstein-11825, 2025-02-28:
Looking a bit more at this picture. Would it be more natural to place the chamber temp box below bed temp? That way all temps are on top.

RodentVienna, 2025-05-26:
a fork that implemented this:
https://github.com/koenpunt/guppyscreen/commit/98e5de57cc2e601955d34c470615978fc7a22f39

i did not check for validity or quality. just stating it has been done somewhat :)