pellcorp/grumpyscreen #48: last file in the file list not selectable

pasqone, 2025-03-17:
When file lists exceed the screen height, it is impossible to scroll down to the last entry and select it. The scroll bar reaches the end of the list without fully displaying the last entry. It is visible only when overscrolling.

Steps to Reproduce:

Open a file list containing more entries than can be displayed on the screen.
Use the scroll bar to scroll to the end of the list.
Observe that the last entry is not fully visible and cannot be selected.

Expected Behavior:

The user should be able to scroll to the end of the file list and select the last entry, regardless of the number of entries.

Actual Behavior:

Scrolling is limited to the physical end of the list, making the last entry inaccessible.

Possible Solution:

Adding a spacer at the end of the file list would extend the scrollable area and resolve the issue.

pellcorp, 2025-03-18:
Can you maybe show a screenshot (or I guess a photo) of what you mean please, so I don't need to go chasing it.

pasqone, 2025-03-18:
it"s not easy to make it clear in a photo so I made a video, I hope it helps:
[https://photos.app.goo.gl/7qjbL1xykJjyHY3JA](url)
In this case it's the Motorplate*.gcode which is not selectable.
Obviously, the workaround is just changing the list order.

pellcorp, 2025-03-26:
ah cool, hopefully can get to it this week

pellcorp, 2025-03-30:
Fixed in https://github.com/pellcorp/guppyscreen/pull/60, Ive no idea if is the correct approach but it works :-)    Will be available soon

closed 2025-03-30 by pellcorp

pasqone, 2025-04-01:
Cooo, gj!

pellcorp, 2025-04-13:
I have pushed a new version of Simple AF with some other fixes, but Simple AF will now forcefully update to the version of guppyscreen with the file listing fix too.