pellcorp/grumpyscreen #68: Fix function get_heaters()

iStarComLab, 2025-05-09:
**Old code:**

```
std::vector<std::string> State::get_heaters() {
  std::lock_guard<std::mutex> guard(lock);
  auto &objects = data["/printer_objs/objects"_json_pointer];
  std::vector<std::string> heaters;
  if (!objects.is_null()) {
    for (auto &o : objects) {
      const std::string &obj_name = o.template get<std::string>();
      if (obj_name == "heater_bed" || obj_name.rfind("heater_generic ", 0) == 0) {
        heaters.push_back(obj_name);
      }
    }
  }
  return heaters;
}
```

**Task:**

The std::vector<std::string> State::get_heaters() function in GuppyScreen should be modified to return only the second part of the name (in this case, chamber) for heaters with a name starting with heater_generic (e.g. heater_generic chamber). This is necessary so that GuppyScreen generates the correct G-code command SET_HEATER_TEMPERATURE HEATER=chamber TARGET={temperature}, avoiding the Malformed command 'SET_HEATER_TEMPERATURE HEATER=heater_generic chamber TARGET={temperature}' error.

**Necessary change**

The function should be changed so that for heaters heater_generic <name> it returns only <name> (e.g. chamber). This will allow GuppyScreen to generate the correct SET_HEATER_TEMPERATURE HEATER=chamber TARGET={temperature} command, which will be correctly processed by Klipper if there is a [heater_generic chamber] section in klipper_config.cfg.

iStarComLab, 2025-05-09:
{
  "color": "blue",
  "controllable": true,
  "display_name": "Chamber",
  "id": "heater_generic chamber"
}

pellcorp, 2025-05-09:
Turns out it was very easy:

```
diff --git a/src/sensor_container.cpp b/src/sensor_container.cpp
index 9202549..aea9184 100644
--- a/src/sensor_container.cpp
+++ b/src/sensor_container.cpp
@@ -1,6 +1,6 @@
 #include "sensor_container.h"
 #include "spdlog/spdlog.h"
-
+#include "utils.h"
 #include <string>
 
 SensorContainer::SensorContainer(KWebSocketClient &c,
@@ -143,7 +143,8 @@ void SensorContainer::handle_edit(lv_event_t *e) {
   if (lv_event_get_code(e) == LV_EVENT_CLICKED) {
     spdlog::trace("sensor callback this {}, {}, {}", id, fmt::ptr(this), fmt::ptr(&numpad));
     numpad.set_callback([this](double v) {
-      ws.gcode_script(fmt::format("SET_HEATER_TEMPERATURE HEATER={} TARGET={}", id, v));
+      std::string heater_name = KUtils::get_obj_name(id);
+      ws.gcode_script(fmt::format("SET_HEATER_TEMPERATURE HEATER={} TARGET={}", heater_name, v));
     });
     numpad.foreground_reset();
   }
```

pellcorp, 2025-05-09:
Will be releasing soon

closed 2025-05-09 by pellcorp